jks-prv/Beagle_SDR_GPS #90: Request - Modification to format of user log file

G8JNJ, 2017-03-08:
Hi John,

I'd like to be able to more easily export the user log file that is generated with the mu command.

At the moment its format is like this

Mar  3 10:10:03 kiwisdr kiwid: 2d:00:27:04 012.  1     343.00 kHz amn z6  "G8JNJ" ::ffff:109.154.135.121 Camden, United Kingdom (LEAVING after 0:01:41)

Is there some way to parse this so that I can extract

Date / Time / Frequency / Mode / Zoom / User / time on frequency so that it can be extracted and loaded into an external  spreadsheet.

At the moment it seems (to me) to be quite difficult to do this as there are no suitable delimiters or standard character spacings that something like Excel can use to split out the required fields.

If it's not easy to parse the existing format,  could some additional  delimiters be added to the generated log file so that it's easier to extract the required data ?

Something like this using semi colons.

Mar  3;10:10:03;kiwisdr kiwid: 2d:00:27:04 012.  1;343.00;kHz;amn;z6;"G8JNJ" ::ffff:109.154.135.121 Camden, United Kingdom;LEAVING after;0:01:41

Thanks,

Martin - G8JNJ

ka7u, 2017-03-20:
Martin,
I played with this a bit tonight. mu is an alias defined in .bashrc - mu="make users"
When I call mu from the command line I get this: 
root@kiwisdr:~# mu
make: ***No rule to make target 'users'.  Stop.

So obviously, I have no clue how to us mu.  So taking a more traditional route, I move into 
root@kiwisdr:/var/log# 
and run:
root@kiwisdr:/var/log# cat user* |grep -i leaving |grep -Fv "192.168.6."
The last statement is to not display my own IP connections, and I get some of this:

> Mar 19 06:23:18 kiwisdr kiwid: 7d:22:50:31 0123 0     3733.00 kHz lsb z10 "3860" ::ffff:47.33.190.220 Bandon, Oregon, USA (LEAVING after 0:41:55)
> Mar 19 06:23:57 kiwisdr kiwid: 7d:22:51:10 0123  1     670.00 kHz  am z0  "::ffff:80.12.55.32" France (LEAVING after 0:01:27)
> Oct 20 02:55:58 kiwisdr kiwid: 1:09:32 0...  1    7180.00 kHz lsb z0  "::ffff:70.212.40.57" Idaho, USA (LEAVING after 0:01:29)
> Oct 20 03:00:21 kiwisdr kiwid: 0:04:04 0...  1    7180.00 kHz lsb z0  "::ffff:70.212.40.57" Idaho, USA (LEAVING after 0:01:22)

It should be fairly simple to parse this output in a spreadsheet, except the kiwisdr kiwid: data set is not uniform in length and has no delimiter to reference. In fact the kiwid: data seems to add in extra spaces for no apparent reason.  So to parse this data, it would help if John could/would add a delimiter or at least establish a fixed character count for the kiwid: 

If he did that, I think you would have a file that could be quickly parsed either with grep and sed, or in a spreadsheet.
Ron - KA7U

DLee-AU, 2017-03-20:
Ron, 

You need to be in the following directory  ~/Beagle_SDR_GPS as root for mu to work.

root@kiwisdr:~/Beagle_SDR_GPS

Regards,

Dustin 
VK4FLEE

G8JNJ, 2017-03-20:
Hi Ron,

> It should be fairly simple to parse this output in a spreadsheet, except the kiwisdr kiwid: data set is not uniform in length and has no delimiter to reference. In fact the kiwid: data seems to add [...]

That's the point I was trying to make.

The current formatting makes this very difficult to do.

Regards,

Martin - G8JNJ

ka7u, 2017-03-21:
Martin,
This command syntax will get you close to what you want:

> root@kiwisdr:/var/log# cat user.log |grep -i leaving |grep -Fv "192.168.6." |awk '{print substr($0,1,16) substr($0,50)}'

It could work perfectly if the 1st day since an upgrade used 0d or some such to correct the column count:

> Mar 20 18:13:12 kiwisdr kiwid: 23:27:25 0.2. 0    14076.00 kHz usb z6  "::ffff:178.34.45.238" Krasnodar, Russia (LEAVING after 1:30:30)
> Mar 20 18:30:55 kiwisdr kiwid: 23:45:08 012.  1    8764.00 kHz usb z7  "::ffff:85.71.10.67" Prague, Czech Republic (LEAVING after 0:03:42)
> Mar 20 18:52:20 kiwisdr kiwid: **1d**:00:06:33 0.2.  1   11175.00 kHz usb z5  "mdmonitor" ::ffff:68.134.198.189 Glen Burnie, Maryland, USA (LEAVING after 0:16:47)
> Mar 20 19:44:42 kiwisdr kiwid: **1d**:00:58:55 012. 0    14076.00 kHz usb z6  "::ffff:178.34.45.238" Krasnodar, Russia (LEAVING after 1:30:28)

I suppose I could figure out an if/then/else script to check the appropriate column for the "d" and then drop it and the character before it, prior to parsing the rest of the string in the output to get the desired result, but I'll need to work on bash scripting to do that. 

I wonder if John is on vacation?
Ron - KA7U

closed 2017-06-13 by jks-prv