jks-prv/Beagle_SDR_GPS #91: WSPR - strong stations are spotted on "sideband" frequencies also...

ON5KQ, 2017-03-12:
I am not sure, whether it is a bug, or I simply missed a proper adjustment.
At least I have not found any kiwiSDR adjustment to prevent the following problem in WSPR rx-mode:

- If there is a strong station reported (usually stns with positive db-value) this station is also reported on the left and right side of the main frequency about 20db lower...
It looks like an amplifier, which is overdriven, producing IM-products  (IM3, IM5) on the left and right side of the main frequency. When I use original WSPR-X program with a different receiver on same band, even extremly strong stations are NOT shown 3 or even 5 times as if every strong station is overdriving their final stages...
My feeling is, that the distortion products are created within kiwiSDR, as I cannot see them with a different receiver.
The problem is very annoying, as the database of WSPR is fed with somewhere "FALSE" data....

It seems a bug in the kiwiSDR software to me...

Ulli, ON5KQ

jks-prv, 2017-03-12:
Hi Ulli,
Yes, this is a known problem and listed in the main bug list: http://kiwisdr.com/bugs/
We think it is due to improper filtering ahead of decimation from the audio sample rate to the sample rate used by the WSPR code (375 Hz).

jks-prv, 2017-05-10:
This Forum post talks about the current workaround in place:
http://valentfx.com/forums/#/discussion/comment/1614
The root problem is still to be understood so I'm labeling this "workaround" instead of "fixed".

jks-prv, 2017-06-03:
Fixed in v1.91

closed 2017-06-03 by jks-prv

ON5KQ, 2017-06-04:
Excellent John!
WSPR mode strong signal handling very much improved - I also think that decoding speed is faster now, especially when many strong signals are in the WSPR passband (often the case here in Europe)

Many thanks for your effort!

Ulli, ON5KQ

jks-prv, 2017-06-04:
Hi Ulli,

Okay, good to hear. Sorry this took so long to resolve. Oddly, it was while working with DSP issues with the new FAX decoder that I realized my mistake with decimation in the WSPR decoder. I’m still learning about all this DSP stuff. It can be complicated sometimes. And I’m not very good at higher-math.

Cheers,
John