jks-prv/Beagle_SDR_GPS #92: Request for enhancement - Notch filter

G8JNJ, 2017-03-16:
Hi John,

I haven't seen this one on the existing list, but I may have missed it.

Would it be possible to add a notch filter ?

Having used the KiWi for some time now, it's about the only major feature I miss in comparison to using a 'normal' receiver.

It would be useful when monitoring stations like NDB's so that strong carriers could be notched within the CW passband.

Regards,

Martin - G8JNJ

jks-prv, 2017-03-16:
It's there (user section: "Add an LMS carrier notch filter") but that list is so long now it's hard to find anything, lol.

G8JNJ, 2017-03-16:
Hi John,

Ah OK, Sorry I missed that.

Regards,

Martin - G8JNJ

jks-prv, 2021-02-27:
We have two LMS-based autonotch filters now plus an autonotch effect from the "spectral noise reduction" filter. See audio tab, "filter" menu. None of these are manually frequency tunable, but they work okay.

closed 2021-02-27 by jks-prv