jks-prv/Beagle_SDR_GPS #93: WSPR suggestions

ON5KQ, 2017-04-11:
Hi - I would like to suggest 2 improvements for WSPR mode:
- is there a chance to use the decoder of K9AN programmed in C ? It seems, that this implementation from K9AN needs much less resources as former original code...
- is there a possiblility to run a predifined number of rx-channels (to be set by admin panel) as WSPR receiver, WITHOUT connection as a client necessary? It means that in admin panel the admin choose to use for example one rx channel for receiving and spotting 40m WSPR and one other channel for receiving and spotting WSPR stations on 80m  WITH NO FURTHER CLIENT connection necessary - so these two channels are only following the admin panel. In that case of coarse only max 2 other "normal"  client rx channels would be left... To stop wspr-receiving one needs to connect the admin panel again...
It means that this function is only for administrator...

WSPR receiving performance is still not very good - I also do not understand the reason of this function as currently implemented, because just using the kiwiSDR audiostream normal USB rx-mode via virtual cable into original WSJT-x program decodes much better! So why doing WSPR receiving - it only make sense, if there is no client connection necessary for it - may be I am missing something ? 
I am using kiwiSDR as a fully remote rx on a far remote rx-location.... no easy access... (remote spectrum analyser, remote rx on different location, etc...)

Many thanks for your continious software support

Ulli, ON5KQ

jks-prv, 2017-04-11:
Hi Ulli,

1) The WSPR extension does use K9AN's decoder written in C as the basis for the code. In fact the latest version of the K9AN code from WSPR-X was integrated about two months ago.

2) This feature has been suggested many times. It is on the wish/bug list. It's a little complicated to implement so I'm not sure when it might happen.

Performance: I have not spent any time comparing decodes from the WSPR extension against Kiwi audio VAC'd to WSPR-X. It shouldn't be that much different since it is using the same K9AN decoder software. Unless there is something wrong in the implementation. There does seem to be a problem with images of strong signals that is not understood.

Why is there a WSPR extension at all? It was chosen as a proof-of-concept that a significant signal decoding function could be written as an extension and integrated into the Kiwi philosophy of doing everything in a browser window. No OS-dependent software installs, no VACs etc. People seem to like the idea and I've had requests to add more digital modes (e.g. PSK31, FSK).

jks-prv, 2017-04-11:
One thing about the latest K9AN decoder. It is much faster than the last version. As a result when it is run on a PC there are now settings available that make it try much harder to pull weak signals out of the noise. This is possible because a PC runs so much faster than than the Beagle. There is much more processing power to spend time decoding weak signals. For example the amount of effort spent by the Fano decoder is a setting that is set much higher on PC WSPR-X. Otherwise on the Beagle it takes too long.

There is even a two-pass process. I'm not sure if this is enabled by default in WSPR-X. It decodes the strong signals first. Then subtracts them out of the received spectrum. Then makes another pass trying to find weaker signals. This may be more important in Europe where the strong signals density seems to be greater. This two-pass process is not possible with the Kiwi WSPR because of how the FFTs are currently implemented.

ON5KQ, 2017-04-11:
John, many thanks for this detailed answer... I am not a programmer and do not know the details of the code. Of coarse I have seen the improvement with the " new" decoder in the WSPR function of KiwiSDR... it was indeed a big improvement. Now as you explain some details to the decoder, I can imagine, that density plays a role and the powerful two-step decoding! Here in Europe the 200Hz WSPR window is completely packed in the evening - especially main bands, 40-80m! In the spectrum you only see one single red, 200Hz wide stream ! Obviously under such busy condition the extra processing power of the PC makes the difference to pick a station every 3 (!) Hz....

ON5KQ, 2017-04-18:
Congratulations on the recent improvements with WSPR (vers 1.73 now) - it makes certainly a difference...

jks-prv, 2017-04-18:
It does seem to help. More work is still needed though. Every time I watch it process signals that fail to decode the first time I think of better ways the remainder of the two minutes might be used to re-process the signals using more decoder effort. You obviously want to limit the amount of time spent re-processing the signals that will never have a chance. But other than signal strength you don't have a lot of initial information with which to make that decision.

closed 2017-06-13 by jks-prv