jks-prv/Beagle_SDR_GPS #94: Feature request: comma and dot as frequency decimal separator in url link

OH1KK, 2017-04-15:
It would be nice if KiwiSDR would accept both comma and dot as frequency separator

http://sdr.vy.fi/?f=3699.2

and

http://sdr.vy.fi/?f=3699,2

should result identical behaviour.

This can be achieved by changing Beagle_SDR_GPS/web/openwebrx/openwebrx.js parse function around row 113 to

```
        s = 'f'; if (q[s]) {
                var p = new RegExp('([0-9.,]*)([^&#z]*)?z?([0-9]*)').exec(q[s]);
                if (p[1]) p[1]=p[1].replace(',', '.');
                if (p[1]) override_freq = parseFloat(p[1]);
                if (p[2]) override_mode = p[2];
                if (p[3]) override_zoom = p[3];
        }
```

jks-prv, 2017-04-15:
Sorry, I keep forgetting about this issue. Just fixed it for the "f=" case (available in the next release). But needs to be fixed in a lot of other places, e.g. admin page fields.

OH1KK, 2019-03-18:
This has been fixed so closing bug. 

Tested with v1.274 and both comma and dot works.

closed 2019-03-18 by OH1KK