jks-prv/Beagle_SDR_GPS #95: GPS Constellation display in web IF

elafargue, 2017-04-18:
It would be awesome to be able to display the current GPS almanach on the admin interface - I designed a nice SVG a _long_ time ago to do this, which you can find at https://github.com/openpilot/OpenPilot/blob/master/artwork/GpsGadget/gpsEarth.svg (open source license, I'm the author), which is very easy to use in HTML... I lack time to learn the UI and contribute, but if you want to use this artwork, you definitely have my blessing!

jks-prv, 2017-04-18:
That's a great idea. I'll have to look and see how simple it is to get the location data in a form that makes it easy to map. You could even plot a little tracking history to give some idea where the sat is heading and when the signal might be "lost" over the horizon. There are lots of sat trackers that seem to do this.

elafargue, 2017-04-18:
NMEA sentences can be an inspiration - for instance the GSV sentence which is easy to use, or the ALM sentence, more complete but more arcane. See http://www.gpsinformation.org/dale/nmea.htm#ALM for details.

jks-prv, 2017-04-18:
Unfortunately there's no NMEA generation in the GPS firmware we use. Since this is a software-defined GPS receiver, and not a complete GPS chipset solution, you don't get all those bells and whistles.

elafargue, 2017-04-19:
Oh sorry: I meant that the syntax of those NMEA sentences is a good
inspiration as to how you can format the data to send to the front-end to
display the almanach/constellation. Basically a list of
PRN/Azimuth/Elevation for the sats in view, or more sophisticated info for
the almanach. But you totally don't have to stick to that format, of course
:)

jks-prv, 2017-09-29:
Things have been pretty difficult lately, so I decided to take a break from the usual development and have a little fun.

So the admin GPS page in v1.130 now has an az/el display using your background image. Thanks very much for that!

I recently got my hp z3801a gps clock working again and I use the control software from GM4JJJ that includes an az/el display. So I was able to check the positions against that. Boy, it was _not_ easy to add az/el capability to the existing GPS code. I had to learn all about ECEF vs ECI coordinate systems etc. All of this was new to me. But really interesting stuff. I've always wanted to know more about it.

closed 2017-09-29 by jks-prv