jks-prv/Beagle_SDR_GPS #96: "Mute" should stop the AUD websocket stream

elafargue, 2017-04-27:
For situations with low network bandwidth, would it make sense to stop sending audio frames when the "mute" button is pressed? This would be a good way to save bandwidth.

ka7u, 2017-04-29:
I use the mute button to silence WSPR when using that extension. If the audio frames stopped what would happen to the active WSPR extension?
Ron - KA7U

jks-prv, 2017-05-03:
Stopping the audio stream on mute would not effect WSPR because WSPR processes the audio IQ data on the Beagle and sends updates to the WSPR waterfall etc. separately. But some extensions, like the timecode extension I've been working on, do use the audio stream already delivered to the browser. So you'd have to detect that situation and prevent the audio stream from being stopped.

It should also be possible to stop the waterfall stream for when you're just listening.