jks-prv/Beagle_SDR_GPS #101: Wrong current users count.

lazywalker, 2017-06-04:
With V1.91, kiwisdr shows 'server is too busy right now (4 users max)' while only 3 users online. Sdr.hu shows  (3 / 4 users) in the mean time, which is correct. 

It was fine before upgrade, bug?

jks-prv, 2017-06-04:
This is the second report of this problem. But it doesn't occur on any of my Kiwis here or a few on sdr.hu that I've tested. Do you have any more information about this? Does it occur on a particular public Kiwi I can try?

dev-zzo, 2017-06-05:
As a side note, this could've been an artifact from my ugly experimental scripts. I've implemented the proper command sequence so now the script instance is accounted as a proper user. I hope this didn't cause much inconvenience. :-) Let's see if the problem persists.

lazywalker, 2017-06-05:
It's okay now by reboot the service, will keep watching.
Thanks

jks-prv, 2017-06-05:
Or if you've ever changed the "number of channels not requiring a password even if password set" entry on the admin security page then there might be a residual bug with that.

jks-prv, 2017-06-08:
Dmitry fixed his code, but it's also a bug that the current API allows "invisible" connections (for the case of code other than the KiwiSDR browser client connecting to a Kiwi server). So I'll call this an actual bug to be fixed.

jks-prv, 2019-01-18:
Andras has also now fixed sdr.hu to correctly show the "max available" channel count as sent by the Kiwi. It used to just be fixed at a value of 4. Also he added the nice feature of moving Kiwis that have all their channels occupied to the bottom of the list and partially greyed-out.