jks-prv/Beagle_SDR_GPS #103: How to control GPS-reception to improve performance

ON5KQ, 2017-06-13:
Dear John,
this is more a question rather than a issue.
My shack in in the basement and I have problems to get good and steady GPS reception. In the administration panel of kiwiSDR we can see the status of the GPS receiver. Currently at best, I have 2-3 good channels, mostly only 1 and often no GPS good channel. I believe my antenna situation in the basement even at a small window to outside is not sufficient...

What does all the status data mean, which is shown in the Admin-panel ?
What data needs to be improved, when trying to get better GPS performance.
I tried to put the (supplied) active GPS antenna outside, via a 20m long Aircell7 cable (high quality 7mm foam 50Ohm coax). However with several adapters in between to change from SMA to PL (coax cable) and back to SMA for the active GPS antenna there is most likely all RF-energy lost... before it reaches the kiwiSDR ;-(

Can you advice for GPS performance improvement for my situation being in the basement shack ?

How many steady channels I would need to achieve a very stable rx-frequency regular corrected from the GPS ?

I recently relocated my KiwiSDR to a new location for less noise , which seem to help. I need to improve reception below 1.5Mhz though, but I am already building additional antennas...

Thanks for advice on GPS.

Greetings from Belgium,

Ulli, ON5KQ

jks-prv, 2017-06-13:
Hi Ulli. You need a minimum of 4 channels for a GPS fix. This also seems to be sufficient to set the SDR clock as accurately as you'd ever want (i.e. a more accurate fix from more sats doesn't improve the clock correction). The GPS correction is mostly fixing the clock offset due to ambient temperature. And then the offset due to manufacturing tolerance.

The GPS display was discussed in this forum post: http://valentfx.com/forums/#/discussion/comment/1402

If you absolutely have to run a longer antenna cable you could try using an active GPS "splitter" that has some gain to overcome the cable losses. Put the splitter after the 3m of cable from the puck antenna. Make sure it can operate on the 3V from the Kiwi, and passes the 3V onto the puck antenna. Alternatively you could just buy a higher-performance GPS antenna. You would still want it to be an active antenna for sufficient gain. But a better antenna element than what the little puck antenna has.

ON5KQ, 2017-06-14:
Many thanks for answer, John,

"....Alternatively, and maybe more simply, you could just buy a higher-performance GPS antenna. ..."

**What is higher performance ?** 
**Do you have spec's of the little GPS antenna , which was supplied with the kiwiSDR ?** 
Then I can compare, what is offered additionally somewhere (Garmin?), and judge whether it will be any better...

Unfortunately active spitters are not cheap - for what is offered, I can easily buy another kiwiSDR of two....

Ulli, ON5KQ

jks-prv, 2017-06-14:
Since you've added to the forum post I mentioned (http://valentfx.com/forums/#/discussion/comment/1402) I'll add my replies there.