jks-prv/Beagle_SDR_GPS #108: Admin panel - Backup does not work !

ON5KQ, 2017-08-07:
Does the backup funtion from the admin panel really work ?

I tried to backup the file system with this admin functionality, but no success !
In the monitor window some information appears, but soon the process stops/hangs for hours...

I then stopped the process and reboot the beagle and try again. The proces is unfortunatly not really repeatable, The only what I can say is, that the backup process stops again, but with different messages, on a different moment 

Finally I feel, trying to backup the file system is may be more dangerous to completely reset the software, than securing the work of several month editing the station labels... There is no information what really happens and when it is expected to be ready. The monitor window shows sometimes some information, sometimes some other information, sometimes no information...
And for sure, there is no backup, unfortunately!

Is there a secure and proven process for non-software Linux experts to take complete backups of the current kiwiSDR status, being able to easily retrieve that status, in case a fatal error occurs and the kiwiSDR software needs to be re-installed ???

Thanks for advice!

Ulli, ON5KQ

jks-prv, 2017-08-07:
Ulli,

I was just able to run a backup from the admin page in v1.109 without problems. It took 3 minutes, 50 seconds for a class 10 sd card. It is possible your sd card isn't working for some reason. This seems to be a very common problem. Can you try another card?

You can run the backup shell script manually outside the Kiwi server to take it out of the equation just to be sure:
1) ssh into the Beagle.
2) "cdp"
3) "cd tools"
4) "lsblk"
You should see your card as "mmcblk1" in the lsblk output:
```
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0boot0 179:8    0    2M  1 disk 
mmcblk0boot1 179:16   0    2M  1 disk 
mmcblk0      179:0    0  3.7G  0 disk 
└─mmcblk0p1  179:1    0  3.7G  0 part /
mmcblk1      179:24   0  7.4G  0 disk 
└─mmcblk1p1  179:25   0  908M  0 part 

```
4) "./kiwiSDR-make-microSD-flasher-from-eMMC.sh"

ON5KQ, 2017-08-07:
Hi John,
I already thought, that writing complete back-up from kiwiSDR should be best, because retrieving it by just booting from that SD-card with reset everything to the latest status... the only thing was, I didn't work...

I need to buy a new SD-card and see what happens... probably the card is not good, for some reason... will let you know result.

Thanks for giving me an idea of backup-time.... I already thought, that waiting 2hours with no action happening, doesn't look like normal.....;-(

Ulli, ON5KQ

closed 2017-08-07 by jks-prv

jks-prv, 2017-08-08:
I sent out a couple of Kiwi boards to people yesterday. When I tried to backup the most recent server version to the sd cards provided by Seeed with the boards I got write errors and had to use other known-good sd cards I had purchased separately (SanDisk 8GB SDHC Class 10). So it looks like Seeed may be sending out bad sd cards. I will discuss this with them. Other cards from Seeed I've had no problems with. They seem to use no-name 4GB class 6 cards, probably to keep the cost down.

ON5KQ, 2017-08-09:
Indeed it works very smoothly ! All it needed was a better SD card. I bought a new one Sandisk Ultra 16Gb, "up to 80MB/s"....
After 2min25sec the backup was complete and says WORKED in green ...

Question: To retrieve this backup, I only need to boot from this card ? What happens if I would have used a non-empty card for the backup ? Will it be erased and overwritten with backup data ?
From the text in the window of the admin tab "BACKUP", it seems the  SD-card will be formatted, so completly erased and rewritten with the backup,,, correct ?

I am no software expert...

Ulli, ON5KQ

jks-prv, 2017-08-09:
Ulli,
When you made the original backup that failed. Were you using the sd card supplied by Seeed or another one? I'd like to know if this was another case of Seeed sending out a bad sd card.

Yes, the backup erases the previous contents of the sd card. And all you do to restore the Beagle on-board file system is to boot from the sd card by having it inserted during a power on. You'll see the "back and forth" LED pattern when the copy is occurring between cd card and Beagle file system.

ON5KQ, 2017-08-09:
John,
at first I used the SD-card, which was supplied with the beagle bone, (it is a black one with 4Gb). Then, after your first answer, I bought a different card from the local shop here. They offer Sandisk and I thought to give this card a try as mentioned above - it is a red/grey one with Sandisk Ultra label on the little card - it worked very smooth - my backup worked immediately.... 2min25sec...

jks-prv, 2017-08-09:
Okay, so the Seeed card was bad. This is what I wanted to know. So I think Seeed has a real problem here and they need to fix it. Seeed is supposed to be creating these backup cards at the factory in the same way you create a backup: by using a backup function on the "/mfg" page which is very similar to the admin backup tab. So if the card were bad it should have failed at the factory and never have been shipped.