jks-prv/Beagle_SDR_GPS #109: Enhancement: Decoder for DRM Digital Radio Mondiale

hubertmeier, 2017-08-11:
Since the KiwiSDR is a shorrt wave receiver, the decoding of DRM Digital Radio is missing. I don't know if its possible to write a decoder without a license or proprietary software, such a decoder would be nice ...

jks-prv, 2017-08-13:
This would be very, very difficult. Even if open source code is available (Dream decoder?) where would it run that has enough compute cycles? The Beagle is pretty well saturated. Getting the code run as javascript inside the browser would be a huge challenge.

Someday it might be possible to run Dream as a separate program on the PC running the browser. Then feed it IQ samples from the Kiwi. Currently the Kiwi doesn't supply enough channel bandwidth to do this. This is probably the best way to go.

closed 2017-10-04 by jks-prv

jks-prv, 2017-10-30:
This will be fixed in v1.139, at least for using the external Dream application via VAC (virtual audio cable) software. Please see: http://valentfx.com/forums/#/discussion/899/drm-demod-yes-external-app-dream-now-working