jks-prv/Beagle_SDR_GPS #110: JT65

Sergeapm, 2017-08-16:
It possibly  to do?
Decoder JT65 into KiwiSDR?
Thanks

jks-prv, 2017-08-16:
It's probably as much work as the WSPR extension. Which means a lot. Not sure about the processing requirements which might be the only difficulty.

Sergeapm, 2017-08-16:
OK   Understand  ,,,  but I am sure  decoders  like HF FAX  WSPR   CW  JT65  and so on  will  be only  to do more popular  your  nice project  KiwiSDR

jks-prv, 2017-08-16:
Agree, but there is only so much I can do. Already working 12/7. The full bug/wish list is here: kiwisdr.com/bugs

Sergeapm, 2017-08-16:
of course  and many thanks for you hard work  all  very good  73!

YO3IBZ, 2020-06-04:
@jks-prv if they ( https://github.com/jketterl/openwebrx ) can decode JT65, FT8, etc. why we can't?

Thank's,
Aurel.

jks-prv, 2020-06-04:
OpenWebRX has little to do with KiwiSDR these days. They are written in different languages, have different server back-ends etc.

I have not looked to see how they did it. But remember that the server-side of OpenWebRX runs on a fully-featured PC where there is a lot more processing power (server-side = place where you plug your RTL-SDR into). They could be running a WSJT-X to do the FT8 decoding that is only modestly modified to interface to the OpenWebRX user interface.

The Kiwi doesn't have this luxury. The Beagle has limited processing power and real-time constraints. Despite this I was able to get the core parts of the WSPR decoder working after much modification to the code from WSJT. It took a long time to do this.

And I don't have the time right now for another long project. DRM took me many, many months. And I'm not personally interested in JT65 and FT8 that much.

YO3IBZ, 2020-06-05:
@jks-prv yes BB has a hardware limitation, i understand and i appreciate your hard work!

Thank you for your reply!
Aurel.

jks-prv, 2023-04-16:
Partially fixed, as we have FT8/FT4 decoding now.