jks-prv/Beagle_SDR_GPS #115: KiwiSDR antenna switch extension init after restart failed

dl2sba, 2017-09-30:
I found one issue with this extension.

After restart of the Kiwi server (for example after FW update) the extension isn't working. 
You have to reboot the beagle bone to get it working.

Is this a problem of the antenna switch extension or of the extension handling in general?

jks-prv, 2017-10-01:
You should open an issue with the antenna switch project, otherwise he will never see it: https://github.com/OH1KK/KiwiSDR-antenna-switch-extension

But please be more specific than "isn't working". As developers this kind of report is not helpful. We don't know what to be looking for or what the conditions are.

closed 2017-10-01 by dl2sba

dl2sba, 2017-10-28:
Fixed 

https://github.com/OH1KK/KiwiSDR-antenna-switch-extension/commit/bea1d060fe210a079d680be462728427872e18b8