jks-prv/Beagle_SDR_GPS #117: v1.131 cause Antenna switch extension works unexpectedly

lazywalker, 2017-10-02:
With v1.131, these two function(see links below) aways return false, that makes we can't deny antenna switching/mixing. Any changes has been made with cfg_int()?

https://github.com/OH1KK/KiwiSDR-antenna-switch-extension/blob/529a843168c44e4927c5156a7e925c8ee5a3f7c8/extensions/ant_switch/extensions/ant_switch/ant_switch.c#L78
https://github.com/OH1KK/KiwiSDR-antenna-switch-extension/blob/529a843168c44e4927c5156a7e925c8ee5a3f7c8/extensions/ant_switch/extensions/ant_switch/ant_switch.c#L86

jks-prv, 2017-10-02:
I just pushed v1.132 that fixes this problem. Restart your Kiwi to update.
Thanks.

lazywalker, 2017-10-02:
Thanks John, such a quick fix

closed 2017-10-02 by lazywalker

jks-prv, 2017-10-02:
I can't believe I made such a stupid mistake. We're lucky it didn't cause any serious problems. However there is now a little bit of corruption in everyone's kiwi.json configuration file (harmless I think) and I'm going to have to write some repair code to fix it.