jks-prv/Beagle_SDR_GPS #118: Forth

jackobian, 2017-10-06:
How to understand  asm code which is forth language

jks-prv, 2017-10-06:
The asm code is not Forth. The asm code runs on the embedded CPU (eCPU) in the FPGA. The eCPU is a very simple 32-bit, stack-based CPU. I know Andrew Holme's description talks about how the instruction set and architecture of the eCPU was influenced by Forth primitives (see http://www.aholme.co.uk/GPS/Main.htm) But there is nothing special about the eCPU.

Unfortunately there is no comprehensive documentation about the eCPU. It is best to study the Kiwi eCPU source code in e_cpu/kiwi.asm to see how each instruction works. Most instructions end in a comment that show the effect on the data stack which is helpful in understanding what the instruction does. You can also study the Verilog code of the eCPU itself in verilog/cpu.v although that requires understanding Verilog. However, there is a C-language simulation of the cpu.v Verilog code in my standalone GPS project that pre-dated the Kiwi: http://www.jks.com/sdgps/sdgps.html

One of the most important jobs of the eCPU is helping to move data around the FPGA under program control. There are special eCPU instructions and data buses connected to the eCPU to help with this task. This customizes the eCPU from being a purely general-purpose processor. The instructions rd/wrevtN (read/write event), rd/wrregN (read/write I/O register) and rdbitN (read/shift/or single input bit) are in this category.

If you are desperate to know the instruction set functioning in detail because you are using the eCPU for a project I could write-up a document. But I don't want to do this unless there is a genuine need. It's good for you to figure this out on your own. You'll learn a LOT. That's what I did!

jackobian, 2017-10-07:
Ok Sir . Thanks A lot I will try it by my self first.

jackobian, 2018-03-16:
Sir, How do u build and compile .asm file?? I have tried masm.32 by  .bat file whih is "
@echo off
path C:\masm32\bin
ml GPS44.asm
if errorlevel == 1 goto end
link /TINY %GPS44,,nul,,,
del %GPS44.obj
rem del %GPS44.lst
:end
"

jks-prv, 2018-03-17:
The gps.asm file in the sdgps distribution (or kiwi.asm if you're using the Beagle_GPS repo) is assembled with a custom assembler, **not** masm.32 You need a system that can compile a .c file to build the custom assembler. Then run the .asm file through it to generate the various output files. The assembler is in asm/asm.c (sdgps) or e_cpu/asm/asm.c (Beagle_GPS) plus some header files. The Makefile in those directories shows how to build everything. I use a Linux or OS X system to do this. I cannot tell you what to do on Windows, sorry.

jks-prv, 2018-03-18:
Oh, but I forgot: Andrew's original code **did** use Microsoft's MASM macro assembler. See the section "Assembly language" on his website http://www.aholme.co.uk/GPS/Main.htm

But since I don't use Windows and wanted an assembler that I had source code for I just wrote my own. I did it in the C language so it would be very portable on Linux/Mac systems. But that also meant changing the source of the gps.asm (kiwi.asm) assembly code to my own format which is not compatible with MASM.

Having source to the assembler I could add special features that made the software easier to maintain. Like single definition files that the assembler processes into header files for C and header files for Verilog. That makes it easier to make a single definition change and have the C code and FPGA code get updated at the same time.