jks-prv/Beagle_SDR_GPS #125: Visual DCF77 decoder

SopaXorzTaker, 2017-10-30:
In my opinion, it would be nice to add a feature to decode DCF77 (and the like) signals.
The GPS synchronization is way more accurate, so this decoder can just be a novelty, maybe a visual comparison between GPS and DCF time.

jks-prv, 2017-10-30:
Hi. I started work on a general time-station timecode decoder extension a while back. But it has not received much attention lately. I got WWVB and TDF phase-mode decoding working but nothing yet for the other stations. You're correct -- it's more of a novelty. But still fun.

jks-prv, 2020-11-17:
The time-station timecode decoder extension has been released now. It needs more work, and needs to support some missing stations and modes.

closed 2020-11-17 by jks-prv