jks-prv/Beagle_SDR_GPS #126: GPS FRONT END

jackobian, 2017-11-01:
Is there any way to use front end from market such as "MAX2769"instead of andrew's one, and how to do that

jks-prv, 2017-11-01:
This is an interesting question. First of all, remember that in the Kiwi design we replaced Andrew's discrete analog front-end with a Skyworks SE4150L chip.

We had considered the Maxim part during the design of the Kiwi board. It is more versatile than the Skyworks part (e.g. it can receive the greater bandwidth of Galileo transmissions). But it is also more expensive and the required chip programming information is a little difficult to determine (it is not fully publicly documented). Since we were only interested in the legacy Navstar (USA) system it was just easier to use the Skyworks part.

If what you are asking is: "can I replace the Skyworks chip with the MAX2769 on the Kiwi board so I can use Andrew's code as currently running on the Kiwi Beagle and FPGA", that question has several considerations. You'd have to build a little daughter board for the MAX, cut the traces on the Kiwi board from the Skyworks (GPS output data and clock) and splice in the MAX. If you can't get the MAX to run at the same 16.368 MHz clock frequency of the Skyworks then you have to change the constant in the gps.h file and recompile. Then you somehow have to program the MAX. I think I remember it has an SPI port. So you could maybe add it to the existing Beagle-to-FPGA SPI with an appropriate enable signal and use the Kiwi SPI code to drive it.

I would have to know more about what exactly you want to do that requires the MAX chip.

jackobian, 2017-11-01:
what i mean use the same Skyworks SE4150L chip with same frac7 board of andrew holme ,just change front-end to market chip like"Skyworks SE4150L chip Or  MAX2769 "

jks-prv, 2017-11-01:
If you are going to use the frac7 board there are a few problems. Frac7 produces the 1575.42 MHz LO for the analog front-end mixer. Then the analog IF signal is sampled at 10 MHz to produce the GPS data stream. The SE4150L takes 16.368 MHz from a TCXO and creates the 1575.42 MHz LO internally. It cannot accept an external LO. There is no need for the LO generated by the DDS part of the frac7. You would just use the frac7 as an FPGA development system for the GPS Verilog. The 16.368 MHz from the SE4150L would be input to the FPGA in place of the 10 MHz. You would need to change the 10 MHz constant in the C++ code to 16.368 MHz and recompile.

What might be easier is to use a commercial FPGA development system instead of making a frac7 board. See here for my pre-KiwiSDR project where I use a commercial XESS XuLA2 FPGA board and a homemade SE4150L front-end: http://www.jks.com/sdgps/sdgps.html

Or you could use a KiwiSDR board and remove the SDR parts of the code to just get the GPS capability. I should probably make this a configuration option of the Kiwi code. Several people have asked about this. They only want a system to experiment with GPS and don't care about the SDR part of KiwiSDR.

jackobian, 2017-11-03:
I have built frac7, but i have facing problem with front end. So i need to buy front end chip from market. i google  about that and found MAX2769 and SE4150L  and there is also zarlink 2010, i do not know which one is best to use with the same frac7 of Andrew's

jks-prv, 2017-11-03:
I would just use the SE4150L. I can't remember if you can run the MAX without having to program it over the SPI which is a complication. I think the Zarlink 2010 was a correlator (acquisition) chip, not a GPS front-end. It is not even available anymore I believe. Anyway, it doesn't matter because the correlation FFT in Andrew's code is doing the acquisition.

jackobian, 2017-11-04:
GP2010 is  Zarlink  GPS Receiver RF Front End. And i can not find schematic of your 2013 old version "24-Sep-2013"

jks-prv, 2017-11-04:
Yes, you are correct. I was thinking about the GP2021 correlator chip. The GP2010 is indeed a front-end. But I still think the SE4150L is the simplest solution.

I never published a schematic for the "breadboard" GPS project using the XuLA2 board (Xilinx Spartan 6 FPGA) because it was so simple. The wiring I used for the SE4150L front-end chip is identical to what's used in the Kiwi: http://kiwisdr.com/docs/KiwiSDR/kiwi.schematic.pdf

Look at the first photo under the "24-Sep-2013" entry in http://www.jks.com/sdgps/sdgps.html It's just 4 wires from the FPGA to SE4150L: power, ground and GPS clock and data. The SE4150L outputs two signals from its ADC, a sign and magnitude. I only use the sign bit. The magnitude bit is ignored.

The 5 wires at bottom right are a standard SPI interface to the Beagle: clk, miso, mosi and a couple of enable lines. The rest of the wires are power/ground and debug/LED wiring. The FPGA is programmed using the USB port on the board using the XuLA tools. On the Kiwi the SPI port does two jobs. It is connected to the FPGA programming pins and after programming those same pins become hooked up to the SPI logic in the Kiwi Verilog. I think frac7 does something similar.

You can also look at the breadboard project Verilog files to understand how the FPGA side is connected. The file sdgps/verilog/top.ucf has the mapping between the pins of the Spartan-6 and net names used in sdgps/verilog/top.v There should be something very similar for Andrew's frac7 board although he's using a Spartan-3 I believe.

jks-prv, 2017-11-13:
There is a recent forum thread that talks about these issues: http://valentfx.com/forums/#/discussion/853/is-it-possible-to-transform-kiwi-sdr-on-xilinx-zynq-fpga-a-single-chip-solution-for-gps-only

jackobian, 2018-03-22:
For front end you only use 1 output of front end chip, why you do not use sign and mag of front end?

jks-prv, 2018-03-22:
In Andrew's original design (http://www.aholme.co.uk/GPS/Main.htm) his front-end is built out of discrete components and a hard limiter effectively becomes his 1-bit ADC. This has a nice effect on the Verilog code. Because the data path is only 1-bit wide all multipliers, like in the mixers, can simply be an XOR gate. You couldn't do that if using 2-bit data, i.e. sign and magnitude output of the SE4150L front-end chip used on the Kiwi. So I only use the sign bit and ignore magnitude. Although it is wired up to the FPGA for future experimentation.

You do lose some sensitivity by not digitizing the signal with greater sampling resolution. I verified this recently while working with Galileo signals in a data file of samples taken from a SE4150L chip. If I ran the acquisition algorithm on the 2-bit data the satellite S/N was greater than if I only used the sign bit.

jackobian, 2018-03-22:
Thanks for reply, i know if i use 2-bit s/r will increase, i am trying to figure out how to do that ?

jks-prv, 2018-03-22:
Well, there are a number of issues to consider. I'll mention a few. Basically you are widening all the data paths in Andrew's drawings that are one bit to two bits. For the tracking circuit at least. Whether you do this for the acquisition sample memory can be a separate step.

Consider this Verilog from demod.v (format slightly different from the original):
```
    wire [3:0] lo_sin = 4'b1100;
    wire [3:0] lo_cos = 4'b0110;

    wire LO_I = lo_sin[lo_phase[31:30]];
    wire LO_Q = lo_cos[lo_phase[31:30]];

        // Mixers
        reg sample, cg_e, cg_p, cg_l;
        reg die, dqe, dip, dqp, dil, dql;
        die <= sample^cg_e^LO_I; dqe <= sample^cg_e^LO_Q;
        dip <= sample^cg_p^LO_I; dqp <= sample^cg_p^LO_Q;
        dil <= sample^cg_l^LO_I; dql <= sample^cg_l^LO_Q;

```
So the LO I/Q oscillator output (LO_I, LO_Q) is mixed with with the input sample (1-bit) and early, prompt or late outputs of the code generator (cg_e, cg_p, cg_l) very easily by using XOR gates. You'll have to change all those to be 2-bit multipliers and then figure out if you want to handle rounding from the 4-bit result. Then there is the LO NCO. LO_I and LO_Q (again, 1-bit wide) are very economically generated from 4-bit lookup tables (lo_sin, lo_cos) using the top 2-bits of the LO NCO phase register. You have to change that to generate two bits or sin and cos to go into your mixer multipliers.

Then there are the integrators (accumulators). They have a certain bit width so they don't overflow. They are read and cleared at each code epoch. Which is 1ms (1023 code chips @ 1.023 MHz) for L1 C/A and 4ms (4092 chips @ 1.023 MHz) for Galileo E1B. With a GPS sampling clock of 16.368 MHz, 1ms is 16368 samples (14-bits) for L1 C/A and 65472 (16-bits) samples for E1B. It turns out you need an extra bit. But even with that L1 C/A only requires 15-bits. And that is good because when the integrated values are read by the e_cpu (green part in Andrew's diagram) most of the code assumes 16-bit values max.

This changes when you start integrating 16368 2-bit samples. The width will need to be greater than 16-bits and suddenly a lot of things become more complicated. Now it turns out I've already solved this problem with my recent Galileo work. Because as you can see above E1B requires 17-bits (16+1). But that code isn't going to be released for maybe a few months. For example, I had to add a new multiply instruction to the e_cpu. It currently has a 16x16 -> 32-bit signed multiplier. I had to add a separate 18x18 -> 36-bit (sign extended to 64-bit) multiplier. I couldn't combine the two because the 64-bit result leaves two words on the stack instead of just one, so it isn't backward compatible. So lots of little things like that to consider.

jackobian, 2018-03-22:
Ok Sir, That was very enough know for me. I will try and see what i can do my self

jackobian, 2018-03-31:
if i us 10MHz instead of 16.368 at SE4150L, is it possible to do that? and is there any thing except Fs should be 10MH and FS_I=10000000,

jks-prv, 2018-03-31:
Unfortunately the SE4150L has a fixed L1 PLL feedback divider so that the input clock has to be fixed at 16.368 MHz. You can move that around a little bit if you want an IF other than 4.092 MHz, but not very much. You can't run it at 10 MHz. (data sheet: http://www.skyworksinc.com/uploads/documents/202445A.pdf)

I have Galileo working now, which means I have 18 and 20-bit versions of the integrator code working. But the position solutions it produces are not accurate (they are off by several hundred meters or more). And there are some other problems with acquisition speed that I don't understand. So I can't release this in the Beagle_SDR_GPS (KiwiSDR) repo just yet.

Maybe I should release it in the Beagle_SDR repo only. That would be some extra work. I would also like to integrate the FPGA C language simulation code from the sdgps project into Beagle_SDR so I can finally get rid of sdgps and stop having two different versions of the same thing.

jackobian, 2018-04-01:
Ok sir, i think it is possible to do that depending on Fif=Fs-mod(Fs,4.093)

jks-prv, 2018-04-01:
Looking carefully at the data sheet it never really gives max/min limits for the tcxo_in clock signal. It only states typical values for the L1 LO and IF center frequencies based on the small variations in the 16.368 MHz reference frequency. There is also the mention that the internal VCO runs at 2x of L1 and that its PLL has a 200 kHz bandwidth.

Looking at the block diagram notice how the reference connects to the usual phase/frequency detector of the PLL. What would happen if you set the reference to 1/2 of 16.368 MHz? Would it still work? The phase detector might. But the frequency detector might always be pulling the loop to one extreme preventing a lock.

A larger problem occurs because of the new ADC sampling frequency of 8.184 MHz. Now you're just sampling at just 2x above the IF frequency and that's going to give you aliasing problems since the -3 dB bandwidth of the IF bandpass filter is 2.2 MHz. The aliasing issue is why 4x the IF (16.368 MHz) is used in the first place.

hcab14, 2018-04-02:
The group delay for 2MHz bandwidth can be different from the one for nominal bandwidth. Similarly the tracking using only part of the BOC signal might behave differently compared to tracking the full signal. However I am not sure that this will result in a position offset of hundreds of meters.

Does the position offset appear for the combined GPS+Galileo position solution or for Galileo alone compared to GPS alone?

One option would be to release the Galileo code in a separate branch.

jks-prv, 2018-04-02:
I get the positioning errors with Galileo only in addition to Galileo + GPS. The first case is difficult because I don't often get 4 Galileo sats within my limited sky view. I modified Taro Suzuki's GNSS-Radar to accept an az/el mask in addition to its existing elevation mask. So my sky view, that has no reception for az = 0..180, is obtained with http://kiwisdr.com/GNSS-radar/?lat=-37.6&lon=176.2&elemask=0&azmask=180&vis=gps,gal,qzs&up=1

Last night I discovered I was incorrectly applying boc(1,1) to the E1B PRN code. So I was effectively acquiring and tracking using the E1B PRN code alone which has the same waveform as GPS L1 C/A and not the double-peak of boc(1,1). I'm surprised this worked at all. For acquisition I guess the correlation still worked because the two lobes of boc(1,1) and the central peak of the E1B PRN code overlap a bit (see waveform picture here: http://valentfx.com/forums/#/discussion/comment/3483) As you'd expect the acquisition and tracking SNR more than doubled as soon as I fixed this.

But the position error remains when Galileo is included. It isn't always wrong. Sometimes I get a run of perfectly good solutions when Galileo is included. This leads me to believe it's just a bug someplace and not a fundamental problem. It can't be an issue with my fix for the TOW spanning problem. Because if the computed sat receive time was off by a bit time (symbol time really, 4 msec) then the error would be huge. Much larger than a few hundred meters (4 msec is ~1200km). But an error in the chip count would be about right (1 chip = ~1 usec ~300m). But the chip count logic is identical to that used for GPS and wouldn't explain why I get runs of good solutions sometimes.

I thought it might be an issue with the LO or code PLL stability. But the admin GPS page IQ display always looks stable. As stable as for the GPS sats.

jackobian, 2018-04-12:
Sir i can have SE4150L , i have gp2015 front end, how to use it with your source code,

jks-prv, 2018-04-12:
I looked at the gp2015 data sheet. Interesting part. The L1 PLL and ADC sample clocks are separate. But unfortunately I found no spec that said what the range of the sample clock could be. They only mentioned 5.714 MHz. Which is what the correlator chip provides by dividing the 40 MHz clock the gp2015 generates by 7. So if you want to do that then put a synchronous 50% divide-by-7 between pins 17 and 11. You could do that in the FPGA. See here for a hint: https://www.onsemi.com/pub/Collateral/AND8001-D.PDF

Now the question becomes: Can you run the GPS stuff at 5.714 MHz instead of 16.368? I don't know. You'd just have to try it and see. It would probably work. The embedded processor in the FPGA would be running at about 1/3 the speed since it uses the same clock at the GPS sample clock. The embedded code still needs to run the PLL loop correction at 1 kHz per channel (the L1 C/A code period). So you have fewer cycles to get that job done. But it never took that much of the processor to begin with. Latency is probably more of an issue. If all 12 channels need to be serviced at the same time, and you can't get to the last one fast enough, I'm not sure what happens.

jackobian, 2018-04-12:
Sir what i try is to downsample by 10MHz which yeild the sae 4.309MHZ, but nothing i can find, and try to downsample by 16.368MHz, and also nothing appear

jks-prv, 2018-04-12:
I answered this same question over on the Kiwi forums: http://valentfx.com/forums/#/discussion/853/is-it-possible-to-transform-kiwi-sdr-on-xilinx-zynq-fpga-a-single-chip-solution-for-gps-only

closed 2018-05-20 by jks-prv