jks-prv/Beagle_SDR_GPS #129: Why ist there a public key for root auth preinstalled??? jks@jks-new.local

krippendorf, 2017-11-05:
A) What happens if I remove it?
B) Did you forget @jks-prv  to remove in postBuild???
[image]

jks-prv, 2017-11-05:
This is leftover from when there were 10 beta sites long ago. I was constantly ssh and rsync'ing to them and was tired of typing the root password all the time. I just forgot to remove this when the build image was sent to Seeed.

What I can do is add a command in the Makefile of the next release to edit the key out of the file. I can't just trunc or remove the file since users might have added their own keys.

jks-prv, 2017-11-06:
Okay, I have a working fix ready for the next release.