jks-prv/Beagle_SDR_GPS #132: Status Json endpint for Telegraph plugin

krippendorf, 2017-11-12:
This is an enhancement / extension idea, not a bug. 

I've setup a small dashboard for our remote QTH: https://dashboard.superlogical.ch/dashboard/db/technikraum-dashboard?orgId=1&from=now-24h&to=now&refresh=30s

Currently parsing the http://kiwiurl/status page to query the number of connected users

[image]

Now I'd like to have a compact json endpoint summary of the Kiwi status to write a custom telegraph plugin like described e.g. here: https://nessy.info/?p=903

Requirements: 
- Number of connected clients
- Current receiver settings / QRG like shown in the onscreen windows buttom left
- FPGA/CPU/MEM usage 
- GPS status
- Network stas

Basically enything in that box: 
[image]

jks-prv, 2017-11-12:
Good idea. The current format of /status is because that's what's needed by sdr.hu Then other projects like http://rx.linkfanel.net/ and http://ve3sun.com/KiwiSDR/index.php began using it. And I use it for maintaining the sdr.hu backup site kiwisdr.com/public No reason not to add a separate JSON interface.

GeoNomad, 2017-11-20:
A JSON interface would be useful.

Related, it would be nice if all the Kiwis could have a callsign type ID. Non-hams could make up a short identifier. The current "name" becomes more like a comment field and it would be good to identify receivers on saved waterfall images with a short ID like a callsign or anything 10 characters max.  I know there is a hex ID now, but that is not very memorable :)

jks-prv, 2017-11-20:
Non-hams who want to register on sdr.hu (which asks for a call when registering for some reason) or want to allow WSPR uploads have been using the convention of SWL<grid_square_4_or_6>. E.g. SWLRF82CI.

I could add a callsign field to the sdr.hu tab on the admin page instead of it only appearing in the WSPR reporter callsign field.

OH1KK, 2019-03-18:
json interface should also be able to communicate with extensions. Like

````
my.kiwisdr.example/json?ext=ant_switch
````

Json would return status on currently selected antennas. 

```
{
  "extension":"ant_switch",
  "backendfailed": false,
  "denymixing": false,
  "denyswitching": false,
  "currently_selected": [ 1,4 ],
  "available": [
    { "antenna": 1, "description": "Wimo GPM-1500 vertical" },
    { "antenna": 2, "description": "100m wire 10m agl" },
    { "antenna": 3, "description": "RA0SMS mini-whip" },
    { "antenna": 4, "description": "50 ohm dummy load" },
    { "antenna": 5, "description": "Sheet metal rooftop" }
  ]
 }
```

If json would be birectional, so it would be possible to also select antennas using json. This would make possible to build hardware antenna switch controller that shows current antenna status and allows selecting antennas. It could be even some physical arduino box with buttons, leds and lcd.

doccodyblue, 2022-07-27:
This one is from 2019 so I may ask ;) 
Is there already a way to get the frequency and the user of RX0 to RXn? 

I'd like to track my users to find out which frequency range is used the most and where I could maybe improve reception.
I could parse the log on a regular base but picking up the data from a (already implemented) JSON would be much more efficient. 

Maybe this info could be added to /status?

jks-prv, 2022-07-27:
Use /status (local network access only for security reasons). [Discussed here](https://forum.kiwisdr.com/index.php?p=/discussion/comment/7982/#Comment_7982).

doccodyblue, 2022-07-28:
Exactly what I was looking for. 

Thanks for the quick answer.