jks-prv/Beagle_SDR_GPS #141: v1.154 . Antenna switch extension opening extension init wrong

dl2sba, 2018-01-02:
After an update from 1.153 to 1.154 there seems to be an issue with the initialization of the antenna switch panel. It shows "selected antenna: unknown" and "permission: unknown" but the antenna(s) are selected correctly. 

Switching works and then also the correct antenna status is shown.

Closing/reopening the panel again same behaviour.

Any ideas?

jks-prv, 2018-01-02:
I'll have a fix in tomorrow's release.

dl2sba, 2018-01-02:
Tnx

OH1KK, 2018-01-02:
dl2sba: I updated antenna switch extension a moment ago for version v1.154. Could you please try new version? 

https://github.com/OH1KK/KiwiSDR-antenna-switch-extension

actually just one file updated

https://raw.githubusercontent.com/OH1KK/KiwiSDR-antenna-switch-extension/master/extensions/ant_switch/ant_switch.js

Works for me (sdr.vy.fi) but I haven't done a lot of testing yet.

jks-prv, 2018-01-02:
I’ve been cleaning up the Kiwi API and didn’t catch that the antenna switch called the now deprecated visible_block() routine. I’ve been careful to keep previous versions of all the old w3_*() routines backward compatible. There will be a dummy visible_block() added in v1.155 later today. All the code in the antenna switch that calls visible_block() can be removed — there is no need for it anymore (it is all handled automatically now).

dl2sba, 2018-01-03:
@OH1KK 

Sri Kari, 

your fix didn't work - still "unknown" ... if you like, I can stay with 1.154 for a while ...

//Dietmar

OH1KK, 2018-01-04:
dl2sba:  It looks that http://dl2sba.ddns.net:8073/ uses still a old version of ant_switch.js . Update ant_switch.js first. You can check what version your KiwiSDR is running version using internet browser. Open url 

http://dl2sba.ddns.net:8073/extensions/ant_switch/ant_switch.js

If first line says (c) 2017 then it is old version.
If first line says (c) 2018 then it is a new version.

Steps to update

First take a ssh connection to your KiwiSDR ant hen

```
curl https://raw.githubusercontent.com/OH1KK/KiwiSDR-antenna-switch-extension/master/extensions/ant_switch/ant_switch.js > /root/extensions/ant_switch/ant_switch.js
cd /root/Beagle_SDR_GPS/
make stop
make install
make start
```

dl2sba, 2018-01-04:
@OH1KK 
Tnx Kari for your support - now it works as expected!

closed 2018-01-04 by dl2sba