jks-prv/Beagle_SDR_GPS #142: Clicking KiwiSDR contact link in 'stats' causes audio to stop (Firefox)

unixpunk, 2018-01-02:
I didn't try this link before the 1.154 upgrade, but at least on my device, clicking that KiwiSDR contact link, which just opens an new email window from outlook for me, causes audio underrun and it does not recover unless you refresh the page, and then all settings which were in the cookie are gone and it returns to the default freq, etc.  If unable to repro locally, should happen to you with mine: http://kiwi.rfoverride.com:18073

Thanks for all the work on this!  Excellent product!

jks-prv, 2018-01-02:
Yes, this is actually a bug in all browsers and there is no fix that I know of. For some reason clicking on a link which starts an email program stops Javascript.

closed 2018-01-02 by jks-prv

unixpunk, 2018-01-02:
Interesting, maybe a security thing then.  Obviously not a big deal anyway.  Thanks for the info!