jks-prv/Beagle_SDR_GPS #146: GPS increase SPI speed

jackobian, 2018-01-28:
As i see from code of solve() function the main execution time is in loadrelpica() function. i follow it and found that something related to spi which is 8M speed, if i increase that speed to 125M can i get high update rate of gps solution

jks-prv, 2018-01-28:
Where did you see a reference to 8 MHz SPI? Are you looking at my sdgps project sources or the new Beagle_GPS sources? I forget what sdgps used but Beagle_GPS runs the SPI at 48 MHz.

But this will make no difference to the rate of GPS solutions. That is determined by the "TaskSleepMsec(4000);" in SolveTask() of either project. This will limit the GPS solutions to 0.25 Hz (once every 4 seconds maximum). I don't know why Andrew has limited the solution rate in this way. You could try decreasing the delay and see if the quality of the solutions remains the same.

jackobian, 2018-01-28:
Sorry i miss the number it is 48Mhz

jks-prv, 2018-01-28:
It's okay. The questions of solution rate speed improvement and accuracy improvement are very interesting and worthy of investigation.

Sometime, tell us about your progress with porting to the Zynq platform.

closed 2018-05-20 by jks-prv