jks-prv/Beagle_SDR_GPS #149: Save ADC clock calibration for use at startup

ckuethe, 2018-03-07:
Of course it's going to vary based on environment over time, but it would be nice to save the clock calibration periodically, and reload it at startup in case GPS is unavailable

ckuethe, 2018-03-07:
Rather than opening a bunch of issues, it occurs to me that it would be handy to have something resembling a warm start: bootstrap the gps with some estimate of local position, time from NTP, loading an almanac...

https://www.navcen.uscg.gov/?pageName=gpsAlmanacs

jks-prv, 2018-03-07:
1) Every time you make a manual clock calibration the value is stored in the configuration and recalled when the server is next started. When there are good GPS fixes the GPS timing updates the current cal. But the manual adjustment value remains and will be applied at the next restart until the GPS once again takes over.

2) Adding a startup almanac capability to the GPS would be a major headache without much benefit. With a good antenna a cold start has a typical TTFF of under a minute. Two at most. A Kiwi is not an automotive navigation GPS.

closed 2018-03-17 by jks-prv