jks-prv/Beagle_SDR_GPS #152: CloseLoop

jackobian, 2018-03-22:
Sir ,can you please provide me with the theoretical equation of how your costas loop is implemented and its equivalent in your code
;	err64 = ext64(pe-pl:32)..
;	ki.e64 = err64 << (ki=ch_CA_gain[0]:16)
;	new64 = (ch_CA_FREQ:64) + ki.e64
;	(ch_CA_FREQ:64) = new64
;	kp.e64 = ki.e64 << ("ki-kp"=ch_CA_gain[1]:16)
;	nco64 = new64 + kp.e64
;	wrReg nco64:32

jks-prv, 2018-03-22:
This is all Andrew's code, so please have a careful read of his description of the Costas loop in the "Hardware/Software" section of his webpage: http://www.aholme.co.uk/GPS/Main.htm

Funny, I was just going through that code a few days ago as part of getting the Galileo sats working -- refreshing my memory about how the tracking code works. Note that the code (EPL) and LO (Costas) loops use a PI controller (viz https://en.wikipedia.org/wiki/PID_controller) as the loop filter function.

Also, see this discussion for information about using a PI controller as a Costas loop filter when an integrator is involved: https://www.dsprelated.com/showthread/comp.dsp/112658-1.php

So just as mentioned in the article above the PI controller has to update the phase value used to generate the LO frequency. It does this by starting with the error signal which is the product of the prompt I & Q integrator outputs: Iprompt * Qprompt (see diagram on Andrew's website). That error is called "~k*k*Theta" in his diagram and called "error" below. Then that chunk of pseudo-code you show above computes:

LO_new_freq = LO_current_freq + (error * LO_KI_gain)
LO_current_freq = LO_new_freq
LO_new_phase = LO_new_freq + (error * LO_KP_gain)

Where LO_KP_gain and LO_KI_gain are the PI gain constants ("K" means "constant" here) passed in from the C code. Multiplication is done with a left-shift (the gains are expressed as a powers-of-two). The math is 64-bits. The final LO_new_phase is the most significant 32-bits of the 64-bit value. 

As part of the Galileo work I actually rewrote that comment as shown below. Note that for the code loop the error signal is the difference in the early and late integrator powers (magnitudes really) pe-pl or (Iearly*Iearly + Qearly*Qearly) - (Ilate*Ilate + Qlate*Qlate).

;   e64 = ext64(LO:ip*qp or CG:pe-pl)
;   ki = ch_CG/LO_gain[0]
;   eki64 = e64 << ki
;   newF64 = *curF64 + eki64
;   *curF64 = newF64
;   kp_m_ki = ch_CG/LO_gain[1]      // stored as kp-ki to simplify code
;   ekp64 = eki64 << kp_m_ki        // i.e. e64 << kp, assumes kp >= ki
;   nco64 = newF64 + ekp64
;   nco32 = nco64 >> 32    // use most significant 32-bits
;
;   effectively:
;   newF = curF + (err << ki)
;   curF = newF
;   nco = (newF + (err << kp)) >> 32

jks-prv, 2018-03-22:
Sorry for the long replies, but I figure they might help someone else who's looking at this stuff. Also, I'm no expert at any of this. I'm learning about it just like everyone else. And writing it down reminds me which parts I really don't understand very well, lol.

closed 2018-05-20 by jks-prv