jks-prv/Beagle_SDR_GPS #155: Wspr autorun - reporting wrong frequency

ON5KQ, 2018-04-26:
Hi John, thanks for fixing the inactivity timeout so quickly, when in WSPR-autorun mode...

Now I found a new issue with the latest versuib 1.85:
I use to have the frequency  offset in WSPR extension tab set to 1500Hz. However when in WSPR autorun mode is running, the spot-frequency of the stations decoded is reported always 750Hz too high - you can monitor this in the kiwi admin log as well as on WSRPNET.ORG

When you run WSPR decoder in manual mode as a client user, the reported frequency is correct with Vers. 1.85

Ulli, ON5KQ

ON5KQ, 2018-04-26:
Addition: One solution is, to set the frequency offset in WSPR extension to 750Hz instead of 1500Hz. In that case the reported frequency in manual mode and autorun mode are both correct.

Is that the intension to make 750Hz mandatory in the WSPR extension setting? At the moment I cannot oversee the consequences, if the setting is fixed to 750Hz - there might be a reason for being flexible with this setting ... (as offered in the instruction of extension tab of the admin panel)

Ulli, ON5KQ

jks-prv, 2018-04-26:
The intent was that autorun connections would use a bfo set by the WSPR_BFO constant (750 Hz) and user invocations of the WSPR extension would use whatever had been set in the BFO field on the admin page. These should be independent and not effect what offset is reported in a spot to wsprnet.org

I don’t understand how there could be a bug with this, but I’ll take a look..

jks-prv, 2018-04-26:
Okay, I see what's happening. I thought I could store the bfo as a global value, but it needs to be stored per WSPR instance so the fixed bfo value of the autorun instance doesn't conflict with whatever has been set on the admin page for user WSPR instances.

jks-prv, 2018-04-26:
Fixed in v1.186

closed 2018-04-26 by jks-prv