jks-prv/Beagle_SDR_GPS #156: Admin password hidden / dots request

zanco, 2018-05-03:
Hi, I am sometimes sharing my kiwisdr by atv transmission in combination with a converrtor. When I change the converter frequency I have  to change the converter setting as admin but I have to turn off my transmitter because the password is shown in full text. Can e have asterisks or dots instead ?

Thanks,
Ben

hcab14, 2018-05-04:
Hi @jks-prv,

I found something which might solve this problem:
https://stackoverflow.com/questions/17769429/get-input-type-text-to-look-like-type-password

There is no need to switch to https. I have tried it locally and for me it works. It would probably be easy to add a tick box next to the password field to make it visible.

hcab14, 2018-05-04:
The method above works only for browsers using webkit.

But this should work for all browsers:
https://github.com/noppa/text-security

I copied text-security-disc.* files to web/kiwi, then added the following lines to web/kiwi/kiwi.css
```
@font-face {
    font-family: 'text-security-disc';
    src: url('text-security-disc.eot');
    src: url('text-security-disc.eot?#iefix') format('embedded-opentype'),
        url('text-security-disc.woff') format('woff'),
        url('text-security-disc.ttf') format('truetype'),
        url('text-security-disc.svg#text-security') format('svg');
}

input.pw
{
    font-family: text-security-disc;
}
```

and made the password form belong to the class `pw`

jks-prv, 2018-05-04:
I was in the middle of implementing something else, but it was ugly. I will try this instead. Thanks!

jks-prv, 2018-05-12:
Fixed in v1.192

closed 2018-05-12 by jks-prv