jks-prv/Beagle_SDR_GPS #159: User listing shows TDoA usage when in other modes

ringof, 2018-07-11:
Using Chrome Version 67.0.3396.99 (Official Build) (32-bit)
Kiwi is kiwisdr.w1euj.com, Up 9:28:45, v1.200, 4 SDR ch, 12 GPS ch

I've used TDoA on this browser/computer before; today I opened up to listen to AM. I noticed that my entry on the Users list shows TDoA usage, though I am not using that extension. 

Looked at the cookies for kiwisdr.w1euj.com, last_mode is as shown in use.

[image]

Happy to pull up anything else you need for this

ringof, 2018-07-11:
Ah, and now its gone and the operating mode shown is correct. I had closed and opened my browser when looking into this issue. Have done it a couple more times since so I guess the issue cleared.

jks-prv, 2018-07-11:
The delay in updating of the user list is perhaps what you're seeing. While connected with TDoA note which slot your connection is using (e.g. RX0). Then close the tab, and connect again in a new tab without using an extension. You'll still see the RX0 entry and it will take a few moments until it goes away and your new connection appears in another slot (e.g. RX1). The delay exists for reasons having to do with complications detecting when the web socket connection has really gone away.

ringof, 2018-07-16:
Good enough for me!

closed 2018-07-16 by ringof

jks-prv, 2018-07-16:
Not ideal. But then lots of things about this software are not ideal I'm afraid..