jks-prv/Beagle_SDR_GPS #160: F11 - full screen

pmolund, 2018-07-12:
In the recent release the F11 - 'full screen switch' doesn't work unless the cursor is in address field while using Chrome. 

This issues seems to have showed up at the same time as the keyboard arrows was redefined to control frequency. (Which I really love as I can now use my Griffin PowerMate as a VFO wheel for the KiwiSDR). 

Regards,
Per

jks-prv, 2018-07-12:
> 
> In the recent release the F11 - 'full screen switch' doesn't work unless the cursor is in address field while using Chrome.

Do you mean the Kiwi frequency entry box or the browser URL/search field
> This issues seems to have showed up at the same time as the keyboard arrows was redefined to control frequency. (Which I really love as I can now use my Griffin PowerMate as a VFO wheel for the KiwiSDR).

The only Windows machine I have is a laptop. On there F11 is a volume control key. There seems to be no function key for full screen. So I don’t know how I can debug this..

pmolund, 2018-07-13:
Normal behavour is that the F11 key would toogle You to/from full screen mode whereever Your cursor are on the screen as long as the browser is in focus. This is the F11 key without any simultaneous keys, ie. no Fn, Alt or 'alt gr' keys. I assume all laptops would support this, however the Fn+F11 function varies a lot between different laptop models and could be the volume control.
 
In KiwiSDR v1.202 You have to set the cursor in the browser URL field in order to enter full screen mode, while the former v1.153 did not show this behaviour. 
This is not a big problem, but more of a annoyance.

Regards Per

danielekmann, 2018-07-16:
This is very easy to fix, at least in Chrome. Replace:

https://github.com/jks-prv/Beagle_SDR_GPS/blob/de861cc84937f9e47e494edc40659d1ef9ca907b/web/openwebrx/openwebrx.js#L5066

with the following:

```js
      if (k == 'Escape' || k.match(/F[1-9][12]?/)) {
```

jks-prv, 2018-07-16:
I'll add that. Thanks Daniel. I'll see if I can test this on my laptop. Maybe the F-keys used with a modifier key gives you the underlying function event, bypassing the laptop assignment (volume, screen brightness etc.)

danielekmann, 2018-07-16:
Isn't "require Fn to press F-keys" an EFI/BIOS thing?

jks-prv, 2018-07-16:
Probably. I'm not a PC / Windows guy, so I'm not sure.

jks-prv, 2018-07-16:
Okay, fix works on my laptop. Fixed in v1.207

closed 2018-07-16 by jks-prv