jks-prv/Beagle_SDR_GPS #161: Kiwi went slow opening since v1.196

lazywalker, 2018-07-19:
From v1.196 to v1.207, my kiwisdr became very slowly everytime I opened the frontend page, every time, I have to wait over 2 minutes to see it totally load, when the loading finished, it operating just normally. reload the page the long loading came again.

When I rollback to v1.195, everything is fine as it used to do, I have to stay at 195 :(

jks-prv, 2018-07-19:
Two minutes? That's terrible. What kind of device (PC, mobile?), OS and browser?

v1.196 added some new javascript to load to support the Google map features needed by the TDoA extension. But it was less than a 10% increase. We use a newer version of W3.css but it isn't much larger (29k -> 32k). I've tried the latest version on an old iPad and a slow PC laptop and it still works same as before. Have you tried other browsers and flushing cache etc?

lazywalker, 2018-07-20:
i guess i know why, kiwi load all js files including ext's at the site was first visited, since v1.96 the TDoA extension use a googlemaps api js file. Unfortunately, all resource from google maps are unavailable from country China because the reason you know why, the request will be hanged until overtime, that would be 2-4 mins:(

So, if i turn on my vpn, everything is fine because the google resources are ok via vpn, but most ppl inside china don't know how to use this trick so it does really confuse to them.

How about lazy loading of all ext js files? we don't have to load all of them the first time, just load as it needed. that might be make life easier :)

[image]
[image]

jks-prv, 2018-07-20:
I was thinking we already had the maps.googleapis.com script tag in index.html. But that of course was for admin.html, not the regular user interface. I'll add async to both of those for the next release.

unixpunk, 2018-07-20:
Not sure if this will help as a workaround, but if the problem is that DNS or the actual IP connection is timing out, then what if it was faked with a /etc/hosts file entry on the machine with the browser being slow?  Assuming its the browser doing the request, otherwise maybe the kiwi might need the host file entry?  Then the browser will just get a real connection refused or 404 error instead and not have to timeout.  Don't know how it will behave in either scenario but easy enough to test.  Hope that helps!

lazywalker, 2018-07-21:
upgraded to v1.208, flushed all caches and do a clean open, direct connect, very fast now.
great job!

closed 2018-07-21 by lazywalker