jks-prv/Beagle_SDR_GPS #163: Cannot save complete received fax images  #extensions #fax

zanco, 2018-08-17:
Hi, yesterday I moved my KiWiSDR to my work location and added an G5RV for reception as antenna. I noticed the decoder plugins for navtex and fax and enjoy those a lot.

There is a but however. For the Fax extension I cannot find a way to save the complete image. Right mouse button, save as, only saves the part which is at the screen on that moment.

Is there a "hidden option" somewhere which I overlooked ?

Tried to find a paypal button to donate the great software support also. Could not find one...

thanks,
Ben

zanco, 2018-08-17:
Hi, yesterday I moved my KiWiSDR to my work location and added an G5RV for reception as antenna. I noticed the decoder plugins for navtex and fax and enjoy those a lot.

There is a but however. For the Fax extension I cannot find a way to save the complete image. Right mouse button, save as, only saves the part which is at the screen on that moment.

Is there a "hidden option" somewhere which I overlooked ?

Tried to find a paypal button to donate the great software support also. Could not find one...

thanks,
Ben

jks-prv, 2018-08-17:
Hi Ben. I tried the fax extension and was able to record and download a .png file that was over 600 lines "tall". This exceeds the vertical size of the screen window which is about 400 pixels IIRC. So I think it should be working. The saved image did include the bits I saw at the beginning of the image when I started recording that had since scrolled off the top of the screen window.

The fax code doesn't have a limit on how many lines the file can contain, but the .png library routines might. So you may have to start and stop the file recordings between each transmitted image (or a little less often) rather than let it run for long periods of time.

There is no Paypal button. Support comes from the sale of KiwiSDR hardware and the software contributions of our users.

Cheers,
John, ZL/KF6VO

closed 2018-08-19 by jks-prv