jks-prv/Beagle_SDR_GPS #164: Lost connection after unplug-plug

iw2nke, 2018-08-23:
After RJ45's unplug - plug on Kiwi then the server dosen't restart. In some case also the Ethernet will remain down. It is a great issue for remote locations in case of power faults.
Thank you,

Gian Leonardo

jks-prv, 2018-08-23:
So, this is not correct.

First of all, are you using the BeagleBone Green supplied with the KiwiSDR kit or your own Beagle, possibly an older BeagleBone Black? Older BeagleBone Blacks have a known hardware problem where the Ethernet interface does not recover if Linux is running and the cable is disconnected then reconnected. But this is **not** a problem with the currently shipping Kiwi hardware/software.

I have just tried this again myself to verify. I can disconnect and reconnect the Ethernet and everything works fine. The server keeps running. It doesn't need to restart. An active admin connection continues to function after the reconnection. A user connection will recover if the disconnection is for a short period of time, like under 10 seconds. Then the waterfall will "catch up" by quickly displaying the data that has been buffered. If the disconnection is longer than that the Linux networking code will close the connection. But you can simply refresh the browser to get a new connection as the server continues to run.

Something that is an issue with power failures (and remote sites) are power supplies that don't meet the power-up specifications of the Beagle. The power supply ramp-up time (0 to 5V) must be under 50 milliseconds otherwise the Beagle may not turn-on.

iw2nke, 2018-08-24:
Hi John,
 I am using the hardware supplied with KiwiSDR.
The power failure is just an example that can occour.
It is possibile to access to a log after the restart to recover some information?
73,

GL

jks-prv, 2018-08-24:
Yes. After a restart ssh or PuTTY into the Beagle and use the command aliases "cdp" and then "msl" to look at the Kiwi log over the last few restarts. The usual "ms" command only shows log messages from the current run.

There is another possibility. Last year Seeed had problems with the Ethernet PHY chips on the BBG. Early in the BBG's life the PHYs would being overheating and fail. The symptoms were no LEDs on the RJ45 connector or the LEDs were on all the time. I personally had two BBGs fail in this way. But there were never any failures in the manner you describe: failure to reconnect after a re-plug.

But it is possible this is a symptom of a partial PHY failure. You could only rule this out by swapping out the BBG and seeing if the problem stopped.