jks-prv/Beagle_SDR_GPS #166: How to set WF off in command line?

lb5va, 2018-09-21:
Hello.
How can parameters like "wf=off"  or "no_wf" be used togheter with frequency and bandwith settings?
I want to have a bookmark that includes all this on my iPhone.
Today I use like: http://mykiwi:8073/?f=3727/-3000,-50lsb
But how to add "wf=off"?
I can't find out how to do it, and may be it is not possible in todays version?
If that's the case ; can it be added?
73's de Stig LB5VA

jks-prv, 2018-09-21:
Hi Stig. You use the ampersand character "&". So "... 8073/?f=7020usb&no_wf&mute" etc. The use of "?" and "&" is standard URL notation not something Kiwi specific.

lb5va, 2018-09-22:
Hello. Many thanks. I have tried that but not succeeded. I dont know why I did not get it to work.
When i tried now it worked fine with "&".. not "?" 
I must have been typing something wrong then I guess.  (feel stupid)...  :-)
Tanks and 73s

jks-prv, 2018-09-22:
The URL syntax is a little weird. The first character starting the parameters is always "?". After that "&" is used to separate parameters. In addition, when a port number is used in the domain part, a slash must precede the "?". So you have to say "... :8073/?f= ..." and not ":8073?f=". But "mykiwi.com?f= ..." would work fine if your Kiwi happened to be on the default port 80.

closed 2019-01-01 by jks-prv