jks-prv/Beagle_SDR_GPS #179: Enhancement: Persistent admin setting to disable IP location [do not call kiwi_geolocate() in openwebrx.js]

HocusLocus, 2018-12-17:
I need my KiwiSDR to start instantly and operate locally with no Internet activity.  Additionally I use NoScript for whitelist security, and any script that tries to access an out-of-bound host (ipapi.co, extreme-ip-lookup.com. get.geojs.io) is shut down.  This stalls the main thread resulting in a screen with no waterfall, menu or audio.  Other than the call to kiwi_geolocate() KiwiSDR operates as a stand-alone product, so I would be grateful if there was a persistent Admin option in Website/Network that sets a disable flag so the served .js would not call the routine.

I realize that these lookups are non-blocking under 'normal' conditions and should fail after a delay.  But in addition to my NoScript example (which protects if the Internet-connected KSDR is compromised) there are other reasons why someone would need KSDR to operate cleanly without attempts at DNS/SSL/Internet.  Mine is part of a 'prepper' package and if I get hit by the bus that takes the grid down, I'd like the KSDR to work offline without subjecting my heirs to diagnosing the 'empty screen' problem I describe above.  To temporarily address it I had to disable updates (because the bus could hit any time), comment it in openwebrx.js, and remake build/obj/edata_embed.o and build/kiwid.bin since the js is embedded.  I'd like to go back to stock.

THANK YOU ALL for this most amazing device! A building's worth of Cold War SIGINT equipment that fits in a pocket.  Whod'a thunk it, or that I could ever afford it!  ~ KickStarterKiwi#1174.

jks-prv, 2019-01-01:
There's a lot more than geolocation that touches the web at startup. I'll have to explore this a bit further.

HocusLocus, 2019-01-01:
This issue just involves the geolocation performed by the client-side javascript, and a request that an admin setting would serve the client .js that does not try to do it... to prevent any dreaded no-net 'stalled thread/no waterfall/menu/audio' client condition.

I am aware that KSDR unit itself has Internet dependencies that cause no-net timeout delay on startup.  An Internet-connected unit is ready to serve local clients in ~1:30 (one minute 30 sec) of powerup.  With no Internet connection the unit does not accept local connections until ~4:51.  This 3:20 delay is  mostly in the startup scripts.  A full 2:40 of that delay seems to be ntpdate (8 DNS timeouts x 20 sec). kiwid itself accepts connections ~46 sec from starting, including its self-imposed 30 sec background startup delay.  

Best case would be tuning the startup so kiwid is run at the earliest possible moment, but it's not a critical issue until the Internet blows up or someone in a log cabin gets a KSDR.  It may become annoying then.

But even without the 'net it is always ready to go in 5 minutes... not unlike tube warmup.  ;-)  

Thx again, Steven

jks-prv, 2019-01-08:
Fixed in v1.257 (no_geo URL parameter). See CHANGE_LOG.

closed 2019-01-08 by jks-prv