jks-prv/Beagle_SDR_GPS #180: Enhancement: Please enable Firefox keyboard shortcuts down to v47

HocusLocus, 2018-12-29:
|| kiwi_isFirefox() <= 52 ||
 [becomes]
|| kiwi_isFirefox() < 47 ||

All keyboard shortcuts verified working in 47.0 ... this version level is of specific interest because it is one of the last that supported the entire previous generation of XUL plugins and functions under Windows XP.  KiwiSDR audio and waterfall functions work fine except recording SaveAs popup (researching why).

jks-prv, 2018-12-29:
But does the delete key work in the frequency entry box with your v47 FF? I added the v52 limitation after a complaint from a user running Vista with this problem (see here: http://valentfx.com/vanilla/discussion/comment/3868)

HocusLocus, 2018-12-29:
Thanks kindly for your quick response.  I'm running FFv47 on WinXP32bitSP3 (with the limit set to "< 47" as above) and also FFv64 WIN10.  KSDR freq entry behaves the same on both:

With no keys pressed, Freq entry field always has browser text focus and is completely selected.  Keyboard shortcuts are processed without affecting it (unless they change freq of course).

IF the DELETE key is pressed first, browser clears entry field because its field had been focused and all text had been selected. After 3 sec timeout if no digits entered, reverts to last freq.  Accepts digits / . and backspace erases and backs up as expected, 3 sec timeout to commit what had been entered.

IF a digit is pressed first, entry field clears and shows the number entered but with no text selection, so additional digits / . can be appended.  Backspace erases and backs up as expected.  ENTER commits instantly, but there is also a 3 second timeout after which freq is committed.

IF an extension with a text field (such as CW decoder with "threshold") is open and that field has been given focus by clicking in it with the mouse, the freq field loses text focus and selection for the duration of the entry.  If the field is modified and ENTER pressed, text focus/selection to the freq field is restored.  Oddly, in the case where the field is clicked (ex, CW:"threshold") and ENTER is pressed without changing the value, focus/selection to the freq field is NOT restored.  This may be an edge case worthy of study.

The basic freq entry seems to be working as it should... perhaps the problems in Vista/Chrome had to do with some condition where text focus and selection was lost unexpectedly rather than some inherent incompatibility.  I realize that having to maintain text focus to a particular field and keep text ranges selected from moment to moment is walking a browser compatibility tightrope these days.  When I convinced keyboard shortcuts to work I expected to discover countless silly little problems, and found none.  Which means... you've done a stellar job keeping your code as clean and beautiful as possible!

jks-prv, 2019-01-01:
This will go out as part of v1.251 today or tomorrow.

closed 2019-01-01 by jks-prv