jks-prv/Beagle_SDR_GPS #190: Impulse Noise

briskiwi, 2019-01-06:
I suffer from neighbours with EOP adaptors. I am able to null out some with my loop antenna but not all. The noise blanker works when off (away) the station of interest but the noise is superimposed over the audio when 'on station'. Is anyone familiar with the code able to improve it's performance when listening to a station?

jks-prv, 2019-01-06:
"Anyone" is just me. And I'm so buried in other work this is unlikely to come to the top of the queue at any point (see kiwisdr.com/bugs). The only hope is if I were to receive a fully tested, ready-to-go patch from someone. Not a very frequent occurrence unfortunately.

briskiwi, 2019-01-07:
No problem. Thanks John.

jks-prv, 2020-11-17:
In addition to the standard time-gate blanker we now have a second one based on the Wild algorithm. The built-in test pulse generator now has some adjustments to help simulate interference so you can gain experience adjusting the blankers.

closed 2020-11-17 by jks-prv