jks-prv/Beagle_SDR_GPS #221: Adding WFM Mode in Kiwisdr Software

longyucheng123, 2019-03-27:
It is very desirable to add WFM mode to the upgraded kiwisdr software and decode WFM (192KHZ)

hcab14, 2019-03-27:
The maximum available audio/IQ bandwidth is either 12 kHz or 20.25 kHz. Both are much less than what is needed for WFM.

jks-prv, 2019-03-27:
What signal on shortwave (<= 30 MHz) uses 192 kHz WFM? Some of the wider Chinese OTHRs? Or perhaps down converting the FM broadcast band? (~ 88 to 108 MHz in the US) For the later it would be much easier to use an RTL-SDR dongle and OpenWebRX software running on an R-Pi or something.

longyucheng123, 2019-03-27:
I hope to add WFM mode to receive down-converted 88-108 Mhz broadcasts. If we can add WFM mode to kiwisdr software, it will be more perfect.

longyucheng123, 2019-03-27:
I hope to add WFM mode to receive down-converted 88-108 Mhz broadcasts. If we can add WFM mode to kiwisdr software, it will be more perfect.发自我的华为手机-------- 原始邮件 --------主题：Re: [jks-prv/Beagle_SDR_GPS] Adding WFM Mode in Kiwisdr Software (#221)发件人：John Seamons 收件人：jks-prv/Beagle_SDR_GPS 抄送：longyucheng123 ,Author What signal on shortwave (<= 30 MHz) uses 192 kHz WFM? Some of the wider Chinese OTHRs? Or perhaps down converting the FM broadcast band? (~ 88 to 108 MHz in the US) For the later it would be much easier to use an RTL-SDR dongle and OpenWebRX software running on an R-Pi or something.

closed 2019-05-10 by jks-prv