jks-prv/Beagle_SDR_GPS #223: 1.276 update has broken IQ recording

DazDSP, 2019-03-28:
IQ recording mode tested.
File type saved is normally 16 bit PCM Stereo.
File type saved now is 32 bit float Stereo.
Sample rate sounds wrong also, but is still 12kHz as normal.
AM mode tested and appears to work OK: Saves a 16 bit PCM and 12kHz file.

jks-prv, 2019-03-29:
I'm guessing that browser recording (a user contributed bit of code) isn't setting something in the header whose absence is being interpreted differently on Mac vs Windows. Because the change I made in v1.276 definitely fixed the issues on the Mac that now appear to be broken on Windows. No one ever complained about it before because fewer users probably use a Mac..

jks-prv, 2019-03-29:
Yeah, what a mess. v1.277 now sets the .wav header according to this specification: http://soundfile.sapp.org/doc/WaveFormat

closed 2019-03-29 by jks-prv