jks-prv/Beagle_SDR_GPS #237: Changing modes jumps waterfall display

ofadam, 2019-05-08:
When changing the mode (like from AM to SSB), the waterfall re-centers back onto the frequency. It would be great to leave the waterfall display in the exact same place.

ofadam, 2019-05-08:
Some clarification – it will jump the waterfall display if you've scrolled far enough to the side of the original frequency where it's no longer visible on the screen. But sometimes I'd like to change modes while continuing to move away from the original frequency.

jks-prv, 2019-05-10:
So the exact case you're describing is not clear to me. Are you talking about changing modes after the passband (yellow graphic) is no longer visible because you've dragged the waterfall or used the circle-<> icons to move the WF sufficiently far away?

The current behavior is that at the end of any frequency-related change (e.g. entry in frequency box, mode selection, zoom change, WF click, etc.) if the passband _center_ is not visible then the WF is readjusted so the PB center is in the middle of the screen. WF dragging and the pan icons are not considered frequency-related.

ofadam, 2019-05-10:
The former (changing modes after the passband is no longer visible.) For some reason, I wish that the waterfall wouldn't budge when changing modes after the passband has been moved off screen. I'm often listening to USB/LSB, but then decide to move the waterfall to see if there's anything else out there – and often find something in an adjacent broadcast band. Switching to AM then jumps me back to the amateur band.