jks-prv/Beagle_SDR_GPS #238: Some icons not high-DPI ready

ofadam, 2019-05-08:
Some of the buttons are not high-DPI, making them appear pixelated on screens like Apple's MacBook Pro line. Could they be changed to 2x references in the template to make them smooth?

[image]

jks-prv, 2019-05-10:
Yeah these icons were done a long time ago by hand before I started to use Font Awesome for everything.

ofadam, 2019-05-10:
I believe that if you make sure that the source images are 2x the normally required resolution, but then reference it at the required resolution in the code, it should make it high-DPI ready.

ofadam, 2019-05-14:
I've adjusted some of the code in a recent pull request. I'll try to continue the work when able.