jks-prv/Beagle_SDR_GPS #239: Unable to close purple info box on iPhone screen size

ofadam, 2019-05-08:
On an iPhone Xs device using Firefox, the purple "welcome" box can't be retracted after opening. In portrait mode, the close button is hidden at the right and can't be scrolled to. In horizontal mode, the floating bar at the top prevents it from being accessed.

[image]
[image]

jks-prv, 2019-05-09:
That, and some other issues for small-screen devices, will be addressed in today's update (v1.288). See the CHANGE_LOG file.

jks-prv, 2019-05-10:
Fixed in today's v1.288 release.

closed 2019-05-10 by jks-prv

ofadam, 2019-05-11:
Horizontal works perfectly! But portrait orientation still means that the purple box can't be closed.

jks-prv, 2019-05-20:
Not sure I understand. You mean you can't close it after you first open it? (which is true). It should be closed by default in portrait mode.

ofadam, 2019-05-20:
It is indeed closed initially – but is impossible to close once opened. I'm not sure there's a good solution, other than taking out a good chunk of text from the introduction, or moving the close button elsewhere.

jks-prv, 2019-05-20:
Okay, I understand now. I've been working towards a mobile-specific user interface for a long time. But there is a lot of support code required and that's what I'm in the middle of now.