jks-prv/Beagle_SDR_GPS #246: How to turn squelch off on NBFM?

RipperFox, 2019-05-23:
While it's sad that a squelch is only available in the mode NBFM i wonder how to turn it off completely - setting it to 0 doesn't seem to do it, since the text 'squelch' still flashes green and audio chopps out.

jks-prv, 2019-05-24:
The squelch code has always been a problem (I didn't write it). On very quiet frequencies setting the squelch slider to zero will not always hold the squelch open as you've noted. You can improve the situation by adjusting the squelch threshold parameter in the URL. Use something like mykiwi:8073/?sqth=5000 or mykiwi:8073/?some_other_URL_parameters&sqth=5000

The value of sqth defaults to 3000 and you need to raise it. Try between 4000 to 6000. But this also has the effect of raising the slider level at which the squelch will close for a given signal strength. So you can end up with the opposite problem.

The whole squelch feature needs to be revisited at some point.

jks-prv, 2020-11-17:
Fixed in v1.419. When the squelch slider is positioned all the way to the left it is now in an "off" (open) state.

closed 2020-11-17 by jks-prv