jks-prv/Beagle_SDR_GPS #247: Google Analytics support

ofadam, 2019-05-28:
Would there be an easy way to add Google Analytics support to the radio via an admin page field? This would help those with public radios track where people are visiting from.

jks-prv, 2019-05-29:
Already there. Scroll to the bottom of the webpage tab on the admin page.

ofadam, 2019-05-29:
Sure enough! I'm not sure how I missed it.

closed 2019-05-29 by ofadam

jks-prv, 2019-05-29:
Well, the admin page is complicated enough now that it's easy to miss things.

Note that when editing the text box it is saved to the Kiwi configuration only after each return/enter keypress. This behavior is due to the same code being used as for single-line field processing. I need to fix this so live editing causes a save after a short timeout.