jks-prv/Beagle_SDR_GPS #249: No SSTV Picture when bandwith is 20kHz

hubertmeier, 2019-06-06:
Hi,

the SSTV feature is nice but when i set the bandwith of the kiwi 3channels/20kHz there does not appear any image (test image), the field remains grey, at standard (4 channels) it works.

Thank you

jks-prv, 2019-06-06:
In the next release an error message will be shown when using the "test image" button while in 3-channel/20 kHz mode. Due to space constraints it is not possible to include the test image file sampled at 20.250 kHz in addition to the one sampled at 12 kHz.

closed 2019-06-06 by jks-prv

hubertmeier, 2019-06-07:
Thank you for your reply, that means, a real signal should work, only the test file does not work in this mode, that is absolutely okay.

Thank you for your hard work !

jks-prv, 2019-06-07:
Correct, and I have verified that on-air signals work in 3-channel/20 kHz mode.