jks-prv/Beagle_SDR_GPS #256: Scroll bars appearing on "user" panel

ofadam, 2019-06-13:
It looks like the v1.290 update made things better for mobile viewports, but added horizontal scroll bars for desktop users (tested in both Firefox and Chrome.) 

[image]

jks-prv, 2019-06-13:
That's been there for a while. It's to accommodate long lines when e.g. there is an extension is use and/or timeout interval and/or long geolocation name. Should go away when not needed.

ofadam, 2019-06-13:
Could it wrap horizontally to just leave the vertical scroll bars as needed? That would also seemingly help on mobile devices.