jks-prv/Beagle_SDR_GPS #257: SDRAngel websocket connection doesn't work in LAN

RipperFox, 2019-06-14:
SDRAngel (https://github.com/f4exb/sdrangel) supports a connection to KiwiSDR via websocket in it's newest release.

It works quite well when connecting to different hosts from e.g. the sdr.hu list.
It doesn't connect however to my KiwiSDR in my LAN - there's a established websocket, but it get's disconnected somehow (had a look with Wireshark & tcpdump). 
When connecting from outside the LAN from a host on the internet to my Kiwi it works fine.
So I suspect some code in kiwid is the reason for this behavior.

Could you please have a look?  

And btw: Thank you very much for this really awesome project! :)
Got my Kiwi connected via a Wifi dongle - I could write a how-to for advanced users if interested..

jks-prv, 2019-06-15:
It may be a while before I can look at this. There doesn’t seem to be an easy build procedure for Mac which is about all I use. I am currently buried in fixing the fallout of bugs from the v1.290 release.

dish-eme-sdr, 2019-06-15:
OFF TOPIC COMMENT:

RipperFox:
"Got my Kiwi connected via a Wifi dongle - I could write a how-to for advanced users if interested.."

I now have a '-SPARE-' kiwi (for experimentation) and have needed to do this more than once when doing portable direction finding.  I ended up using my RedPitaya which I do have setup to connect via USB WIFI dongle.  I can also configure it to act as an access point which greatly simplifies connecting to things like Android tablets, etc. without having to drag along a router and other network equipment!  I've been thinking for trying to do the same with the kiwi.  Maybe I can speed this along by taking a look at what you have.

I've found a wireless connection is also a great way to eliminate noise from network cables & equipment.

I can contact you directly via your DL6 callsign if you prefer as I don't want to tie up #257 with off topic posts :-^)...

jks-prv:
Sorry for the off topic bandwidth.

and

A great big THANK - YOU for your devotion to this project.  I wish I could help more but my coding skills aren't up to the task.  I'm more of a patch / hacker / experimenter than anything else...

73 for now...

RipperFox, 2019-06-15:
@dish-eme-sdr Little guide here: 
https://gist.github.com/RipperFox/b08f7fb54709f139be756368d37d5de1
Call is dk9sas - got a club mail address, too :)

dish-eme-sdr, 2019-06-15:
Ripperfox:

THANK YOU !!!

VERY similar to config for RedPitaya...

Now to see if I can make it work as an access point as well.

This would allow me to take a loop antenna, kiwi and Android tablet out into the field without having to drag along a router and other network equipment.  Also reduces noise and battery drain :-^)...

I now return control of #257 to it's owner.

Thanks for the OFF-TOPIC bandwidth...

RipperFox, 2019-07-13:
Seems latest SDRAngel can now connect to my Kiwi in LAN - problem was on SDRAngels side I guess..

closed 2019-07-13 by RipperFox