jks-prv/Beagle_SDR_GPS #281: Request: Add CB channel labels

ofadam, 2019-07-20:
Could the CB channel numbers be added to the default labels?

ringof, 2019-08-03:
The labels (stored in the dx.json file) seem to be intended as an example rather than a complete common label scheme. Many KiwiSDR operators have been updating theirs to be specific to their location - my own included. See [my dx.json](https://github.com/ringof/kiwi-dx.json) for a set of US CB channel labels. 

Here's the JSON for US CB Channels:

,[26965.00,"AM","CB:1","",{"SB":1}]
,[26975.00,"AM","CB:2","",{"SB":1}]
,[26985.00,"AM","CB:3","",{"SB":1}]
,[27005.00,"AM","CB:4","",{"SB":1}]
,[27015.00,"AM","CB:5","",{"SB":1}]
,[27025.00,"AM","CB:6","",{"SB":1}]
,[27035.00,"AM","CB:7","",{"SB":1}]
,[27055.00,"AM","CB:8","",{"SB":1}]
,[27065.00,"AM","CB:9","",{"SB":1}]
,[27075.00,"AM","CB:10","",{"SB":1}]
,[27085.00,"AM","CB:11","",{"SB":1}]
,[27105.00,"AM","CB:12","",{"SB":1}]
,[27115.00,"AM","CB:13","",{"SB":1}]
,[27125.00,"AM","CB:14","",{"SB":1}]
,[27135.00,"AM","CB:15","",{"SB":1}]
,[27155.00,"AM","CB:16","",{"SB":1}]
,[27165.00,"AM","CB:17","",{"SB":1}]
,[27175.00,"AM","CB:18","",{"SB":1}]
,[27185.00,"AM","CB:19","",{"SB":1}]
,[27205.00,"AM","CB:20","",{"SB":1}]
,[27215.00,"AM","CB:21","",{"SB":1}]
,[27225.00,"AM","CB:22","",{"SB":1}]
,[27255.00,"AM","CB:23","",{"SB":1}]
,[27235.00,"AM","CB:24","",{"SB":1}]
,[27245.00,"AM","CB:25","",{"SB":1}]
,[27265.00,"AM","CB:26","",{"SB":1}]
,[27275.00,"AM","CB:27","",{"SB":1}]
,[27285.00,"AM","CB:28","",{"SB":1}]
,[27295.00,"AM","CB:29","",{"SB":1}]
,[27305.00,"AM","CB:30","",{"SB":1}]
,[27315.00,"AM","CB:31","",{"SB":1}]
,[27325.00,"AM","CB:32","",{"SB":1}]
,[27335.00,"AM","CB:33","",{"SB":1}]
,[27345.00,"AM","CB:34","",{"SB":1}]
,[27355.00,"AM","CB:35","",{"SB":1}]
,[27365.00,"AM","CB:36","",{"SB":1}]
,[27375.00,"AM","CB:37","",{"SB":1}]
,[27385.00,"AM","CB:38","",{"SB":1}]
,[27395.00,"AM","CB:39","",{"SB":1}]
,[27405.00,"AM","CB:40","",{"SB":1}]

ofadam, 2019-08-21:
Thanks for the help!

closed 2019-08-21 by ofadam