jks-prv/Beagle_SDR_GPS #282: Feature Request: L/R/Both panning. client or server side

HocusLocus, 2019-07-23:
The objective is to be able to run two Kiwi sessions to discrete speakers for any monaural mode except IQ, an Audio panel  control that can set to L/R/Both to optionally put out to one Stereo channel only.  IF it could be done client side with StereoPannerNode that would be great...  or at very least on server side by transmitting a sparse stereo signal.  Thx kindly and great work!

jks-prv, 2019-08-16:
I really like this idea. I didn't know about createStereoPanner() so this was relatively easy to implement. It will be in today's v1.312 release. See the Kiwi forum for more info.

closed 2019-08-16 by jks-prv

HocusLocus, 2019-08-17:
Two thumbs up, works great! That was an easy one for the practical result.  I appreciate your warning that audio is insufficiently synchronized to do some cool stuff but there's a buttload of missing timecode infrastructure and these network and js thread environments are piles of ever changing latency jello on slow machines.  Two comms or one with a MW station is like 2 radios in one to us simple folk. Of course volume and pan should be built in to every browser tab in the 21st Century, but I'm glad KSDR did it first.

jks-prv, 2019-08-18:
I hope it's useful. As usual it took me 10 minutes to add the feature. Then 4 hours to get the UI working, test it on all the browser/platform combinations and do a release. It's a wonder I'm able to add anything new these days.