jks-prv/Beagle_SDR_GPS #288: Suggestion: Integrate/average waterfall if WF-rate is slower than "fast"

helioshk, 2019-08-17:
As far as I understand, if the WF-rate is for example 1 Hz, kiwid will take a spectrum snapshot each second and then add it to the waterfall.

It would be nice if there was some averaging/integrating of all available spectrum snapshots between the updates on the web interface.

That should give a smoother waterfall appearance if WF-rate is slower than the "fast" setting. It should also increase the visual SNR, if values are integrated (since noise averages out and signal adds up).

Thanks for your continuous effort. Appreciate it much. Take care.

jks-prv, 2019-08-23:
Interesting idea.

helioshk, 2019-08-24:
[Not my idea](https://en.wikipedia.org/wiki/Signal_averaging), some SDRs do that already. Gqrx for example. Or signal processing software like argo or spectrum lab.