jks-prv/Beagle_SDR_GPS #293: Max number of connections from same IP address

ofadam, 2019-08-21:
Would it be possible to set a max number of connections from the same IP address? I'm getting folks who are loading all 4 slots to monitor net frequencies, leaving nothing for others. (I've already set max times per listener and IP, but it would be great to limit to one slot per IP.)

QDeathNick, 2019-08-25:
Is it difficult to add support for multiple logins to the system?
It would be great to limit to one slot per login.

hcab14, 2019-08-25:
I cannot see what is wrong with net monitoring. In any case, when you want to use your KiwiSDR yourself you can always kick user connections.

If you have already set a maximum time limit per listener and per IP and a user connects to all four slots he can use them only for up to (time limit)/4 per 24h...

closed 2019-08-28 by jks-prv