jks-prv/Beagle_SDR_GPS #324: CW offset configurable

RightRudder, 2019-11-09:
I wish there was an option on the admin config tab to set the default offset for CW and CWN filters.  They always default to a center frequency of 500hz and I prefer a different tone for listening.  I can drag the upper and lower skirts but next time I connect to my KiwiSDR  from a different machine its back to the default tone.  Could this be put on a wishlist for enhancement please.

jks-prv, 2019-11-15:
Until then, the quickest way to change the passband center is to enter ":pbc" into the frequency box of the control panel, e.g. ":700" for a 700 Hz pbc or ":1k" which is equivalent to ":1000" (something like ":1.2k" works as well). You can add this to a frequency specification as "freq:pbo" e.g. "7020:850". You can also specify the width ":pbo,pbw". See: http://kiwisdr.com/quickstart/index.html#id-freq-entry

RightRudder, 2019-11-15:
Oh yeah I remember reading that during my initial learning on the kiwi but forgot about that.  There is also the URL method and now I realize I can add the PBWand PBC switches to my bookmark for my local kiwi so it will always open in my prefered CW listening mode.  Thanks for the reminder.  It would still be nice to have a config option though when time allows to add that.

Best regards
Joe

jks-prv, 2019-11-23:
Fixed in v1.350

New "Initial CW offset (Hz)" option on admin config tab. Of course this can only be changed by the owner/admin of a Kiwi and so doesn't help a user accessing Kiwis they don't control. That will have to wait for the mythical "user preferences" feature.

closed 2019-11-23 by jks-prv

RightRudder, 2019-11-24:
Thank you John.